ECE477/Hardwear-Design #7: BNO-055 (IMU) VDDIO vs. VDD

mitchellciupak, 2020-10-07:
https://cdn-learn.adafruit.com/assets/assets/000/036/832/original/BST_BNO055_DS000_14.pdf
^ Datasheet for BNO-055 (See page 12 for Electrical Overview and page 98 for I2C connection diagram(our connection))

- Can we connect VDDIO and VDD to the same voltage source? It looks like they are doing it on our breakout (https://www.adafruit.com/product/2472)

mitchellciupak, 2020-10-07:
More docs --> https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/downloads

monacok21, 2020-10-07:
According to the datasheet, VDD has to be between 2.4 to 3.6 and VDDIO has to be between 1.7 and 3.6 so connecting both to 3.3V should be fine.

ckemmet, 2020-10-09:
They didn't seem to have a problem with it at all. I'm closing this.

closed 2020-10-09 by ckemmet