ECE477/Hardwear-Design #17: PCB Layout Reminders

monacok21, 2020-10-08:
1. Decoupling capacitors should be as close as possible to pwr/gnd pins (usually placed on opposite side of ICs)
2. On/Off switch should hang off board
3. Don't ground fill where antenna connections are
4. Orthogonal Routing
5. Layout traces at desired widths (don't layout small then increase trace widths)

mitchellciupak, 2020-10-08:
6. All user pushbuttons should contain a 10-100 ohm resistor in series to discharge current that a user could be carrying.
7. JTAG -> 4 pin connector
8. Remove GPS LED
9. Update Naming Convention
10. Don't use any single pin connections (should have reference to ground)
11. Make sure all connectors are polarized (cannot be put in backward)

mitchellciupak, 2020-10-11:
Completed: 8, 9, 10, 6, 7(?)
Will use in PCB design: 1,2,3,4,5,11

mitchellciupak, 2020-10-11:
Breaking into sperate issues

closed 2020-10-11 by mitchellciupak